OpenBCI/OpenBCI_Cyton_Library #12: Remove delays in firmware

yannrenard, 2016-04-28:
The communication protocol on the firmware side implements many (hardcoded) delays. These delays aren't documented and it is difficult to make timings optimal on the client side without falling in cases where the delays would not be sufficient as compared to firmware expectations

pushtheworldllc, 2016-04-28:
I believe you are referencing:
https://github.com/OpenBCI/OpenBCI_32bit/blob/master/OpenBCI_32bit/OpenBCI_32bit.ino#L79

and 

https://github.com/OpenBCI/OpenBCI_32bit/blob/master/OpenBCI_32bit/OpenBCI_32bit.ino#L508

yannrenard, 2016-04-28:
Those are the long ones... but you could also consider :

https://github.com/OpenBCI/OpenBCI_32bit/blob/master/OpenBCI_32bit/OpenBCI_32bit.ino#L500
https://github.com/OpenBCI/OpenBCI_32bit/blob/master/OpenBCI_32bit/OpenBCI_32bit.ino#L469
https://github.com/OpenBCI/OpenBCI_32bit/blob/master/OpenBCI_32bit/OpenBCI_32bit.ino#L434
and others

These delays should be dealt with at a lower communication layer than the actual protocol implementation

pushtheworldllc, 2016-05-10:
Addressing and testing this now...

closed 2016-05-11 by pushtheworldllc