OpenBCI/OpenBCI_Cyton_Library #13: Make communication protocol machine readable instead of human readable

yannrenard, 2016-04-28:
Right now communication protocol sends (or may not send) replies in plain text, with information being (or not being) present depending on the current status of the board. It would be much easier for client code to be able to parse machine-ready replies formatted in JSON, XML or at least following a machine parseable format, especially including delimiters for begin and end of a reply. The content of the reply should be consistent across different device configuration, e.g. :

{
"version" : 3,
"max-channels" : 16,
"on-board-ads1299" : "0x3E",
"on-daisy-ads1299" : "0x3E",
"has-daisy" : true,
}

(note the 'has-daisy' which would show-up even though daisy is not attached)

pushtheworldllc, 2016-04-28:
this is just a good idea....

pushtheworldllc, 2016-04-28:
@yannrenard what do you think about sending a different soft reset to indicate start with machine readable code. If you sent for example an `m` or something that indicates: `soft_reset` and ack back in JSON

yannrenard, 2016-04-28:
would be awesome ! then make all the calls to send a reply, even though it is a simple 
```json
{
	"status" : "good"
}
```

:)

pushtheworldllc, 2016-05-11:
Look at sending variable names as strings

aj-ptw, 2017-06-29:
This has been done with the wifi shield.

closed 2017-06-29 by aj-ptw

yannrenard, 2017-06-29:
awesome work @aj-ptw