OpenBCI/OpenBCI_Cyton_Library #14: Handshake at connection

yannrenard, 2016-04-28:
At connection time, it would be great to have a handshake to declare the device to client code. This allows to easily scan available ports and figure out which port have an OpenBCI plugged in and have basic information about the capabilities of the device. This handshake may also include a protocol version !

pushtheworldllc, 2016-04-28:
Ok so how is this different then sending a `v`

yannrenard, 2016-04-28:
Because sending 'v' assumes you already handle some sending / parsing.
When you scan for devices instead, you would basically have a fraction of the code separated of the actual driver and have minimal work to deal with, just for the purpose of scanning ; this would be pretty much like identifying the magic bytes or 4CC in files. (MZ for microsoft executables, PK for zip files, #! for scripts etc...)

pushtheworldllc, 2016-05-09:
ok i think i am understanding this now. This would look like a private command that never left the host. Ok i think that is doable.

wjcroft, 2016-05-29:
"This would look like a private command that never left the host. "

AJ, I think you are equating 'host' with the current dongle scheme. Ideally this initial connection information would allow the laptop program to determine several parameters which include information from the OpenBCI mainboard; and allow the laptop program to adjust it's protocol parsing software:

* which mainboard? 8bit, 32bit, or Ganglion
* number of main and aux channels
* revision level of firmware
* etc...

Including @biomurph and @conorrussomanno  here.

Regards, William

pushtheworldllc, 2016-05-29:
@wjcroft the current scheme provides book way to identify an OpenBCI dongle. I have programmed in a way that sending just 0x00 from the PC to Dongle ('Host') will return the channel number of the Dongle.

pushtheworldllc, 2016-08-01:
Proposed solution has been posted here: https://github.com/pushtheworldllc/Docs/blob/master/software/01-OpenBCI_SDK.md#openbci-command-protocol-overview

And can be found in the latest release candidate for the library https://github.com/pushtheworldllc/OpenBCI_32bit_Library/releases/tag/v2.0.0-rc.5

Further (and my favorite) quick status messages as described here https://github.com/pushtheworldllc/Docs/blob/master/software/01-OpenBCI_SDK.md#radio-configuration-commands which allow you to communicate with the radio system and get status messages if asked for them. Since the radio system is the mission critical barrier to success with this system, knowing if your radio system is up or down seems to help a lot in application development.

aj-ptw, 2016-08-31:
I want to start this discussion again and figure out the best way to do this moving forward.

I think the best solution would be to add a new command that is simply a loop back from the board to the driver. So let's say we send a `?` or something and all that does on the board is sends a `Success$$$` message back to the driver.

aj-ptw, 2017-08-10:
This has been completed once and for all with the wifi shield checkout OpenBCI/OpenBCI_Wifi

closed 2017-08-10 by aj-ptw