OpenBCI/OpenBCI_Cyton_Library #21: ADD type for analog sensors

pushtheworldllc, 2016-05-09:
There is sometimes a need to convert aux bytes into numbers, should this be done?

biomurph, 2016-05-09:
Yes,
It is sometimes useful to have the AUX bytes converted without any scale
factor. This should be an option.
So far, we only need to convert the Accelerometer values with a scale
factor.

~~~
Making the world safe for robots

On Mon, May 9, 2016 at 10:24 AM, AJ Keller <notifications@github.com> wrote:

> There is sometimes a need to convert aux bytes into numbers, should this
> be done?
>
> —
> You are receiving this because you are subscribed to this thread.
> Reply to this email directly or view it on GitHub
> <https://github.com/OpenBCI/OpenBCI_32bit_Library/issues/21>
>

pushtheworldllc, 2016-08-01:
This has been completed see @pushtheworldllc master docs here for an explanation of the new possible end bytes! https://github.com/pushtheworldllc/Docs/blob/master/software/02-OpenBCI_Streaming_Data_Format.md#binary-format

closed 2016-08-01 by pushtheworldllc