OpenBCI/OpenBCI_Cyton_Library #22: Interrupt for ADS data ready pin

pushtheworldllc, 2016-05-10:
Check out RFdigital for examples of how to do this.

pushtheworldllc, 2016-05-10:
Seems like a good guide: http://chipkit.net/wp-content/uploads/2016/02/MajenkoTech-AN1132-Working-with-chipKIT-Interrupts.pdf

pushtheworldllc, 2016-05-10:
Another longer guide http://chipkit.net/wp-content/uploads/2015/05/chipKIT_PIC32_Interrupt_Handling.pdf

pushtheworldllc, 2016-05-10:
@biomurph here are some good tutorials

biomurph, 2016-05-12:
Hey! I poked around and figured out how to make interrupts work!
The code here has been tested:
https://github.com/biomurph/OpenBCI_32_Interrupt

wjcroft, 2016-05-29:
Joel, very cool, thanks.

aj-ptw, 2016-08-31:
This has been complete and can be found in all examples here https://github.com/OpenBCI/OpenBCI_32bit_Library/tree/2.0.0

closed 2016-08-31 by aj-ptw