OpenBCI/OpenBCI_Cyton_Library #24: ADD: SD Card variable write length

pushtheworldllc, 2016-05-13:
Be able to send a string command in the form of hours and minutes to at runtime declare a block write size.

yj-xxxiii, 2016-05-13:
Something like that   ,    Andrew ?

 *      - Define the duration of the record session. 
 *        To do this, send an 'hxxhxx>' .
 *        ex : 'h1h>' 	  sets to 1 hour
 *             'h2h30>'	  sets to 2 hours 30 minutes.
 *             'hh5>'     sets to          5 minutes.
 *             'hh99h'	  sets to 1 hour  39 minutes.
 *             'hh125>'   sets to 2 hours  5 minutes.
 *             'hh1234>'  sets to 2 hours 34 minutes!!
 *             'hh12345>' sets to 7 hours 45 minutes!!! The code is minimalist, so stay inside ' xxhxx '
 *         this command echoes 
 *             - the duration formated as 'xxhxx'
 *             - the corresponding closest number of blocks according to the current sample rate and the recording format (hex/bin)

`
case waiting_duration_value 	: 
     if(testChar == '>')
          { 
             SessionSize = (SessionTime*60*(250<<(0b110-Sample_Rate_SD_Only)))/512 ; 	
             SessionSize *= (Bin_format ? 32 : 80) ;
	     Serial0.print(SessionTime/60); 
             Serial0.print("h"); 
             Serial0.print((SessionTime%60)/10); 
             Serial0.println(SessionTime%10);                                 delay(10);
	     Serial0.print(SessionSize); Serial0.println(" blocks");    delay(10);
             command_state = waiting_command_beginning ;
              break ; 
         } 

    if(testChar == 'h') 
        {
          SessionTime *= 60 ;
          break ; 
        }

     if(!(('0' <= testChar) && (testChar <= '9')) )
       { 
        Serial0.println(" Error, do it again"); delay(10); 
        command_state = waiting_command_beginning ;  	break ; 
       }
     else 
      { 
        SessionTime += ((SessionTime%60)*9) + (testChar-'0') ; 
        break ; 
      } 
` (sorry cant format the code... y.j.)

pushtheworldllc, 2016-05-14:
Updated the formatting @yj-xxxiii you can click the "pencil" to see how to work with github markup. 

I think what you wrote is right on the money!