OpenBCI/OpenBCI_Cyton_Library #25: Board ADS running too fast, producing more than 250 samples per second.

pushtheworldllc, 2016-05-14:
@jfrey-xx reported with his boards that they seem to be running too fast.

jfrey-xx, 2016-05-14:
...not a brand new issue, here are my initial concerns: http://openbci.com/forum/index.php?p=/discussion/313/ads1299-sample-rate-drift-or-jitter :p

yj-xxxiii, 2016-05-16:
Even 1% faster !!!
Why does this matter ?

Anyway it is not a surprise since the internal clock accuracy  of the ads1299 is +/-0.5% at 35°C and +/- 2.5% between -40&85°C.

y.j.

wjcroft, 2016-05-29:
Mentioning Joel @biomurph and @conorrussomanno here.

(1) the real issue is the lack of the crystal on the current mainboard; understandable as a cost saving measure in the early days of the first Kickstarter. But much less justifiable at the current pricing.

(2) if the 8 channel board is ever revised, suggestion: put in the traces / pads so an external crystal could be EASILY added with no SMT soldering skills. That way those users that needed the extra accuracy could upgrade on their own.

(3) Ensure that the accuracy of the Ganglion sample rate (256 sps?) is adequate. Optionally have crystal solder-in available as a DIY addon.

Regards, William

pushtheworldllc, 2016-05-29:
Yann Renard allowed me the open vibe driver that has a drift correction scheme. I think the new time synched packets will allow people to easily offline determine the rate of their ADS clock at the time of.

wjcroft, 2016-05-29:
AJ, thanks. I'm aware of the OpenViBE drift correction, if you look through the threads on the forum. Nothing can beat the utility of an accurate sample rate. There is a whole paper there to read by David Hairston, have you seen it?

Accounting for Timing Drift and Variability in Contemporary Electroencepholography Systems
http://www.dtic.mil/dtic/tr/fulltext/u2/a561715.pdf

Our drift is worse than any of the other systems listed there. So may be a concern from a marketing point of view.

yj-xxxiii, 2016-05-30:
Could it help to read a counter of the PIC32 which is crystal driven, and to write this high resolution time stamp in the auxiliary data field? 
y.j.
P..S. : I never tried to study evoked potentials without recording a hard trigger signal on a dedicated channel. This way a 1% drift is not a problem...

pushtheworldllc, 2016-08-01:
Added timestamps to the data packets which will allow you to take record, start time stamping, take the (last time stamp - first time stamp) / samples to find the number of samples captured in comparison with the Driver. Works well if you time sync up with a Driver using new protocol and continually resync every minute or so.

closed 2016-08-01 by pushtheworldllc