OpenBCI/OpenBCI_Cyton_Library #26: ADD:  Variable Sample_Rate

yj-xxxiii, 2016-05-16:
- Set the sample rate to 250, 500, 1000, 2000Hz. 
  To do this, send an '.' followed by a char in '1' to '4' corresponding to 250..2000.
  ex : '.2' sets to 500Hz

`case waiting_freq_value :
if('1' <= testChar && testChar <='4') { 
         Sample_Rate_SD_Only = 0b110-(testChar-'1'); 
         if(EEPROM.read(2)!=Sample_Rate_SD_Only) EEPROM.write(2 , (uint8_t)Sample_Rate_SD_Only); 
        Serial0.println(250<<(0b110-Sample_Rate_SD_Only));  
     }
    command_state = waiting_command_beginning ;  break ;
...
...
case 'N': writeToSDonly = true; OBCI.Sample_Rate = Sample_Rate_SD_Only ;
...
`
In OpenBCI_32_Daisy.cpp
`WREG(CONFIG1,0xB0+Sample_Rate,BOARD_ADS); // tell on-board ADS to output its clk, set the data rate to Sample_Rate
WREG(CONFIG1,0x90+Sample_Rate,BOARD_ADS); // turn off clk output if no daisy present
`
I am still unable to paste the code properly formatted...
y.j.

pushtheworldllc, 2016-06-10:
Just saw another person interested in this. Will be looking to do this soon!

eohan, 2016-06-10:
+1 .. I'd like to measure at 4khz , but 2khz would get good!

aj-ptw, 2016-08-31:
Thanks for the code @yj-xxxiii looking at it now. Was having trouble with setting on the fly but your code should help me lots!

aj-ptw, 2016-08-31:
This will be addressed in 3.0.0.

aj-ptw, 2017-06-17:
Check out the new release! https://github.com/OpenBCI/OpenBCI_32bit_Library/releases/tag/v3.0.0-beta1 for full wifi support

closed 2017-06-17 by aj-ptw