OpenBCI/OpenBCI_Cyton_Library #27: Tracking: Over The Air Programming Failures

pushtheworldllc, 2016-06-15:
This issue shall be used to report over the air programming failures.

pushtheworldllc, 2016-06-15:
## Steps to reproduce error
1. Flashed new radio firmware code to the Device radio (_Board_)
2. Flashed new radio firmware code to the Host radio (_Dongle_)
3. Failed to upload Pic 32 firmware

## Computer System Details
Laptop : Apple MacBook Pro (Retina, 15-inch, Late 2013)
OS Version: OS X El Capitan 10.11.3
Processor: 2.3 GHz Intel Core i7
Memory: 16 GB 1600 MHz DDR3

pushtheworldllc, 2016-07-12:
## Error
Could not erase chip. Failed immediately.

## Steps to reproduce error
1.  Not able to reproduce

## Computer System Details
Laptop: Toshiba
OS Version: Windows 7 64bit
Processor: Intel Core i5 1.7 GHz
Memory (RAM): 8 GB
Poll Time: 87ms

eohan, 2016-07-26:
## Error
Program flash: .................................................################################################## done
Verify flash: .................................................#invalid reply checksum
Read page failed.

## Steps to reproduce error
1. Upload the radio code to device and host: https://github.com/pushtheworldllc/OpenBCI_Radios/releases/tag/v2.0.0-rc.2
2. I get the above error on Arduino IDE when I upload DefaultBoard.ino firmware to the PIC
Firmware code : https://github.com/pushtheworldllc/OpenBCI_32bit_Library/releases/tag/v2.0.0-rc.3
## Computer System Details
Laptop: Dell M 4700
OS Version: Windows 8 Pro 64 bit
Processor: Inter (R) Core (TM) i7-3740QM CPU @ 2.70Ghz
Memory (RAM):  8GB
Poll Time (ms): tried 80 and 85 (us?)

pushtheworldllc, 2016-08-01:
Please see the latest radio release here https://github.com/pushtheworldllc/OpenBCI_Radios/releases/tag/v2.0.0-rc.4 for a proposed patch. Make sure you speed up your `pollTime` with a `F00432` to about 50ms to not get rock speed uploads from previous release candidates. See the `Release Candidate 4` section for notes about the changes.

closed 2017-06-17 by aj-ptw