OpenBCI/OpenBCI_Cyton_Library #39: Add: Impedance testing on board

aj-ptw, 2016-10-01:
Be able to compute impedance on the board and just send the values to the computer to *drastically* reduce the time it takes to measure impedances.

jpdandrade, 2016-11-14:
It would be nice to "automatically"/easily setup the impedance signal from the firmware code. We could define the frequency we wanted for the impedance signal and it would be always on. Even better it would be sending the impedance measurement in the data packet, if possible.

aj-ptw, 2016-11-23:
@jpdandrade thanks for the code upload. 

I notice you do not make a call to start streaming. Are you doing that from the driver?

jpdandrade, 2016-11-23:
@aj-ptw yeah my goal was not to code the "start streaming", although it would be interesting to know how to do it as well.
I just wanted to set the impedance measurement signal to be always "active" and not only when I command it through the GUI. This way I would able to measure the power spectrum in the specific frequency bin to measure impedance.

aj-ptw, 2016-11-23:
And you are not seeing the impedance signal at all? Could you post a screen shot of the signal vs. the other signals. This should work as you described of always being on.

jpdandrade, 2016-11-24:
- After the upload of the code (the one on git) to the board, I have channel 1 ready for impedance measurement:  https://www.dropbox.com/s/jllk7a6hncky0do/IMG_20161124_161233.jpg?dl=0.
- Initialized the OpenBCI_GUI (modified to be able to see the frequencies above 60 Hz in the frequency spectrum plot) and started streaming. Impedance frequency peak is not visible: https://www.dropbox.com/s/xcbmnoaf525z3ex/signal1.png?dl=0
- clicked in the impedance measurement button for channel one and impedance frequency peak appears: https://www.dropbox.com/s/bwksz94njlmfric/signal2.png?dl=0

This was what made me think I wasn't doing it right in the code. I don't know if it's asking too much, but it would be great if you could try it and see how to make it work.
Thanks!

jpdandrade, 2016-12-02:
Just putting more wood in the fire:

1. Not connecting wires to board. Only channel 1 on: 
https://cloud.githubusercontent.com/assets/5629079/20839717/8cec60d0-b8a5-11e6-81bb-62c15133f0d5.png
(I can see peak at 50Hz and 62.5Hz, as disired)

2. Connecting only gray wire to board. Only channel 1 on: 
https://cloud.githubusercontent.com/assets/5629079/20839833/f6899364-b8a5-11e6-9b1b-1e06b176be6e.png
(very noisy. peak at 62.5Hz disapears)

3. Connecting gray, white and black wires to board (they are not touching each other). Only channel 1 on:
https://cloud.githubusercontent.com/assets/5629079/20839930/5f183cdc-b8a6-11e6-9d41-977067a3a0a0.png
(can't see 62.5Hz peak)

4. Connecting gray, white and black wires to each other. Only channel 1 on:
https://cloud.githubusercontent.com/assets/5629079/20840009/afb1014c-b8a6-11e6-8933-660e569afd53.png
(all noise disapears)

5. Connecting gray, white and black wires to each other. All channels on:
https://cloud.githubusercontent.com/assets/5629079/20840080/e5336292-b8a6-11e6-872e-e27a89600084.png
(can see 50 Hzpeak)

It seems that the code on board actually makes us have the Fs/4Hz peak (case 1.), but why does the signal behaves like I show in the other cases? This looks strange to me...
Sorry for long post but I'm trying to be clear