OpenBCI/OpenBCI_Cyton_Library #46: compile fails - Arduino can't find <OpenBCI system library>

hoosierEE, 2017-02-01:
MacOS 10.11.6, Arduino 1.6.5

Following along with the README, I install all the prerequisites and then open one of the OpenBCI examples.  Compilation fails with this mesage:

```
Arduino: 1.6.5 (Mac OS X), Board: "OpenBCI 32"

Using library OBCI32_SD in folder: /Users/ashroyer/Documents/Arduino/libraries/OBCI32_SD (legacy)

/Users/ashroyer/Library/Arduino15/packages/chipKIT/tools/pic32-tools/1.42-pic32gcc/bin/pic32-g++ -c -g -O2 -Wall -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -mprocessor=32MX250F128B -DF_CPU=40000000L -DARDUINO=10605 -D_BOARD_DP32_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino -G1024 -D_USE_USB_FOR_SERIAL_ -I/var/folders/27/w1myz2vn78gflt11gb_hydfr0000gp/T/build8231903212779785058.tmp/sketch -I/Users/ashroyer/Library/Arduino15/packages/chipKIT/hardware/pic32/1.3.1/cores/pic32 -I/Users/ashroyer/Library/Arduino15/packages/chipKIT/hardware/pic32/1.3.1/variants/OpenBCI -I/Users/ashroyer/Documents/Arduino/libraries/OBCI32_SD /var/folders/27/w1myz2vn78gflt11gb_hydfr0000gp/T/build8231903212779785058.tmp/ReadWrite.cpp -o /var/folders/27/w1myz2vn78gflt11gb_hydfr0000gp/T/build8231903212779785058.tmp/ReadWrite.cpp.o 
ReadWrite.pde:24:23: fatal error: OBCI32_SD.h: No such file or directory
compilation terminated.
/Users/ashroyer/Library/Arduino15/packages/chipKIT/tools/pic32-tools/1.42-pic32gcc/bin/pic32-g++ returned 255
Error compiling.
```

I thought it was kind of odd that the SD library is contained in an otherwise empty `OpenBCI_32bit_SD` folder, so I copied `OBCI32_SD` into `~/Documents/Arduino/libraries`

Now I get this error:

```
Arduino: 1.6.5 (Mac OS X), Board: "OpenBCI 32"

Using library OBCI32_SD in folder: /Users/ashroyer/Documents/Arduino/libraries/OBCI32_SD (legacy)

/Users/ashroyer/Library/Arduino15/packages/chipKIT/tools/pic32-tools/1.42-pic32gcc/bin/pic32-g++ -c -g -O2 -Wall -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -mprocessor=32MX250F128B -DF_CPU=40000000L -DARDUINO=10605 -D_BOARD_DP32_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino -G1024 -D_USE_USB_FOR_SERIAL_ -I/var/folders/27/w1myz2vn78gflt11gb_hydfr0000gp/T/build8231903212779785058.tmp/sketch -I/Users/ashroyer/Library/Arduino15/packages/chipKIT/hardware/pic32/1.3.1/cores/pic32 -I/Users/ashroyer/Library/Arduino15/packages/chipKIT/hardware/pic32/1.3.1/variants/OpenBCI -I/Users/ashroyer/Documents/Arduino/libraries/OBCI32_SD /var/folders/27/w1myz2vn78gflt11gb_hydfr0000gp/T/build8231903212779785058.tmp/ReadWrite.cpp -o /var/folders/27/w1myz2vn78gflt11gb_hydfr0000gp/T/build8231903212779785058.tmp/ReadWrite.cpp.o 
In file included from ReadWrite.pde:24:0:
/Users/ashroyer/Documents/Arduino/libraries/OBCI32_SD/OBCI32_SD.h:19:18: fatal error: DSPI.h: No such file or directory
 #include <DSPI.h>
                  ^
compilation terminated.
/Users/ashroyer/Library/Arduino15/packages/chipKIT/tools/pic32-tools/1.42-pic32gcc/bin/pic32-g++ returned 255
Error compiling.
```

Any ideas?

biomurph, 2017-02-01:
I think this is a known problem with Arduino IDE 1.6.5.
Please upgrade to the latest Arduino version and try again.

hoosierEE, 2017-02-01:
Ah, I was following [the OpenBCI_32Bit readme](https://github.com/OpenBCI/OpenBCI_32bit), which says:

> Download Arduino IDE version 1.6.5 software from the Arduino site. Currently, only 1.6.5 works with our software.

Perhaps that readme should be revised. On 1.6.12 it compiles just fine.  Thanks!

closed 2017-02-01 by hoosierEE

biomurph, 2017-02-01:
Oh,

You should not use that code! The 32bit Libraries have been upgraded.
Go here
https://github.com/OpenBCI/OpenBCI_32bit_Library
and find the file DefaultBoard for our produciton code. You can find some other code in the examples there too.

hoosierEE, 2017-02-07:
Nice, thanks!