OpenBCI/OpenBCI_Cyton_Library #51: Cyton writing bad files to sd card

dsteinkraus, 2017-05-27:
I have a Cyton which I've been using for several months, saving data to the sd card. At first, the files were formatted just as the documentation shows they should be. But last week I opened one of the recent files, and found they had switched at some point to containing binary data that makes no sense to me. As an example, a recent file starts with these hex values:

41 10 01 A0 B0 00 01....

The change happened at a definite point in time, and all files since then have been more or less like this. It happens with both Linux and Windows hosts, and both with my own software (which uses OpenBCI_NodeJS to talk to the Cyton) and with the OPENBCI_GUI app. I'm using a Transcend 8GB class 10 card, which I've formatted several times using the SDFormatter app. The files are written to the SD card with the expected names and sizes, they just contain what looks like binary garbage.

Cyton firmware 2.0.0.

wjcroft suggested firmware might be corrupted. Will try reflashing.

aj-ptw, 2017-06-12:
Any update on this?

aj-ptw, 2017-08-10:
I think the SD card may of gotten corrupted, did you try a new card?

aj-ptw, 2017-11-01:
use a sands card

closed 2017-11-01 by aj-ptw