OpenBCI/OpenBCI_Cyton_Library #67: Feature request: User callback function in the DefaultBoard.ino

gerrievanzyl, 2017-08-16:
This suggestion may make the firmware more hackable by users with limited experience.  

To make small changes how/what data is logged, would it be possible to create user function in DefaultBoard.ino that is called just before data is written to serial, wifi and SD card (something like void userDataProcessing() ). This would be setup as a simple call back that is run just before the EEG and accel/aux data is written out.  

The purpose of this callback would be to give users one simple place to make any modifications to the data before it is written out.  This would eliminate the need for users to mess with any of the Library code.

For example: to insert "markers" into the data streams the callback function would allow someone the ability to change the auxData[] values just before they are written out.

Make sense?

aj-ptw, 2017-08-26:
wow this is a cool feature request! I must not be getting notifications by default for this repo. Do you know how to do call backs in arduino?

biomurph, 2017-08-27:
This might be easy to do, but how best to make it usable? We could create an .ino file in the sketch example 'UserProcessing.ino' or so. There could live the userDataProcessing() with default empty brackets. ??

bonebizz21, 2017-09-15:
Thanks again