OpenBCI/OpenBCI_Cyton_Library #68: EEG hiccups when '`' is sent via serial

gerrievanzyl, 2017-08-16:
I have tried the following on 2.0.1 and 3.0.0 dev firmware.

In my attempts to insert markers into the stream I have used the functionality in 2.0.1 to send the  '`' character. When I type this on the keyboard it creates a "hiccup" on the EEG data stream in the GUI plots. See the attached screenshot.  I have not checked whether these artifacts are present in the SD files.  This instability would certainly limit the ability gather data where stimulations are handled by the Cyton board.

I tried this on a modified 3.0.0 dev firmware (modified to expect `n  and insert a marker with value n into auxData[0] ) and it creates the same artifact.

What I did notice was that whenever I sent serial commands to the board, it did miss at least one packet.  Is this just a problem with the GUI filters?  

It is not clear to me why sending 2 chars to the board (with trivial processing on the board side) would mess up timing enough to miss one packet.  Is this this possibly a bug?

[image]

aj-ptw, 2017-08-26:
hi can you please paste your firmware code?
the hiccup you see in the GUI is because the board temporarily stops the stream for some commands, such as setting the channels. Is the ``` taken? Maybe there is another function that is using that already, I can't remember.

gerrievanzyl, 2017-08-28:
AJ,

Firmware code has been pushed.  I am 99.9% sure it is not the FW as the
hiccup is not visible in the SD card data.

I have done more testing and found that the hiccup happens 9 out of 10
times.

Here is some further info:

   - When running from processing I get the message in the console that the
   packet counter skipped from x to x+2
   - The instances when it does not "hiccup" there is no associated message
   that the packet counter skipped
   - I have tried putting in code to sync sending the marker message with
   the incoming packet but this did not help. (The code would hold off sending
   the marker message until it had detected the end of the 33 byte packet)
   - When I disable the notch filters, the missing data is almost not
   visible

gerrievanzyl, 2017-10-31:
I believe this only applicable to BLE connections - it may be because we are pushing the BLE connection hard and any interference with the stream of data results in lost packets.    I am closing this for now.  If I find it is a problem on wifi I will reopen

closed 2017-10-31 by gerrievanzyl