OpenBCI/OpenBCI_Cyton_Library #74: Issue recording at 2kHz on Cyton

NeuralControl, 2017-09-04:
Hello, 
I am trying ask via python to record on sd card at 1kHz, using v3.0.0
I looked around and tried quite a bit but cant get it to work.
My code looks like below, does anyone knows what i'm doing wrong by chance?
i get createfdContiguous fail without softreset, and if i do a soft reset i get back to 250Hz,
Thanks

board.ser.write(b'~6');sleep(1);read() # ask 250Hz
board.ser.write(b'v');sleep(1);read()  # soft reset
board.getSampleRate()                  # check 250.0
board.ser_write(b'a');sleep(1);read()  # record ~14sec
sleep(20)
board.ser.write(b'~4');sleep(1);read() # ask 1kHz
board.ser.write(b'v');sleep(1);read()  # soft reset
board.getSampleRate()                  # check 1kHz
board.ser_write(b'a');sleep(1);read()  # record ~14sec
sleep(20)

aj-ptw, 2017-10-16:
> i get createfdContiguous fail without softreset, and if i do a soft reset i get back to 250Hz,

Can you rephrase?

aj-ptw, 2017-10-16:
I think you are writing too fast too it

NeuralControl, 2017-10-16:
Thanks AJ. I switched to wifi :)