OpenBCI/OpenBCI_Cyton_Library #82: Data not writing to SD card on Cyton with GUI v3.1.0 and v2.2.1 on Mac

aghariba, 2017-11-01:
It initiates and you can see the data stream in the GUI, but no SD file is created.  I'm not having this problem with GUI v2.2.1 on Mac. Board firmware is v3.1.0.

gerrievanzyl, 2017-11-01:
AJ is working on this he has some new versions in development stream that seem to solve this. Give it a day or so.

gerrievanzyl, 2017-11-03:
@aghariba,  @aj-ptw (the main developer of the project at the current time) is super busy on a whole lot of things - I am sure there will be an update to the main branch of the GUI soon.  However, I have a "development" branch that addressed this issue for me.  You can find it here: https://github.com/gerrievanzyl/OpenBCI_GUI/tree/feature/UDPmarkers

The branch you are looking for is the feature/UDPmarkers branch. 

Remember to download the latest Electron Hub for system from here: https://github.com/OpenBCI/OpenBCI_Ganglion_Electron/releases
and save the OpenBCIHub application in the /OpenBCI_GUI/OpenBCI_GUI/data/ folder.

As I said, this is working for me, but I am not in a position to support the code.  Try it, if it works great! if not delete it.

aj-ptw, 2017-11-03:
Oh this explains why I didn’t see this issue here, it’s a GUI issue, my friend above is right, we’ll get this merged ASAP! Sorry!

aj-ptw, 2017-11-03:
Which OS do you have? I'm about to put a new build out

aghariba, 2017-11-03:
I'm on a Mac. Thanks for fixing this so quickly!

closed 2022-10-13 by retiutut