OpenBCI/OpenBCI_Cyton_Library #84: Inaccurate SD Card maximum file writing times

gerrievanzyl, 2017-11-06:
In the OpenBCI GUI you can select the SD file sized based on the expected duration of the trail.

It would appear that these sizes are based off a 8 channel data writing rates at 250Hz.  With 16 channels at 1kHz a 1-Hour file is filled in 7.5 minutes. 

We need to update the file size based on the sample rate and the board type settings.

I can tackle this if required. 

A simple way to implement this would be in the code that allocates the SD space by simply multiplying the number of blocks defined (see below) by 2 for 16-Channel board and 2 or 4 for 500 or 1000Hz sample rate.

#define BLOCK_5MIN  11000
#define BLOCK_15MIN  33000
#define BLOCK_30MIN  66000
#define BLOCK_1HR  131000
#define BLOCK_2HR  261000
#define BLOCK_4HR  521000
#define BLOCK_12HR  1561000
#define BLOCK_24HR  3122000

aj-ptw, 2017-11-07:
we should do this programatically but i 100% agree this needs to be done!

aghariba, 2017-11-08:
I tried to double the BLOCK_24HR to get 24hr 16ch recordings, but then OpenBCI often fails to initialize through the GUI.  Is it necessary to pre-allocate space?  It would be ideal if it can just keep writing until the memory card is full.

gerrievanzyl, 2017-11-08:
@aghariba it looks to me as if the limitation is on the SD library the code uses.  I do not have any background to this, but I know that it is a good strategy in cases like these where we have something like a microcontroller trying to run a file operating system.   I see there are newer SD arduino libraries available, that do not have this limitation, but I do not know how well they perform. 
For now, I propose that we make the small changes and @aj-ptw can evaluate whether it is worth rewriting this functionality.

I will give a shot at making some changes and submitting a pull request

aj-ptw, 2017-11-12:
@gerrievanzyl I think this is a great idea!!

gerrievanzyl, 2018-01-11:
@aghariba were you able to make the changes?

aghariba, 2018-01-11:
I haven't tried to do this yet.

retiutut, 2024-10-09:
closed by #102

closed 2024-10-09 by retiutut