OpenBCI/OpenBCI_Cyton_Library #94: Changing sample rate breaks impedance measurement

bmartin427, 2019-12-05:
Changing the sample rate seems to clobber the LOFF register (0x04) of the ADS1299.  This can be easily reproduced by using the sequence of serial commands `?`, `~6`, and `?` following reset.  The LOFF register has the value 0x02 prior to the sample rate command, and 0x00 afterwards.  This value change configures the ADS1299 for DC lead-off detection instead of 31.2Hz AC, which is not what the GUI expects to use.

This issue is the underlying cause of OpenBCI/OpenBCI_GUI#653.  However, no wifi board is required to reproduce this issue as described above.  Wifi is involved in the original issue only insofar as the GUI only attempts to change the sample rate at startup when using the wifi interface, and not serial.

wjcroft, 2019-12-05:
Brad, thanks very much for tracking this down.

bmartin427, 2019-12-05:
I have confirmed that PR #95 makes the 31.2Hz waveform show up in the GUI using the wifi interface when clicking the 'ohm' icons.